isbeorn/nina #8: Speckle Interferometry Plugin Fails to Install

jaxtonmaez, 2025-06-10:
When downloading NINA in the Plugin Tab, an error occurs where the Plugin Installation fails. That's all the infomration given. Even when manually downloaded and extracted into the proper folder, the plugin does not appear in the installed plugin section. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Plugins'
2. Scroll down to 'Speckle Interferometry'
3. Click 'install'
4. See error

marcblank, 2025-06-10:
What is the error?   Can you post the NINA log?

jaxtonmaez, 2025-06-10:
Good to know, I will get onto the Discord. Heres's an image of the error. No other information is given. 

[image]>

marcblank, 2025-06-10:
Please post a log, as requested.

NINA_Speckle.png (view on web)<https://github.com/user-attachments/assets/2a45cc39-e1ed-4149-be32-e724921ffe0d>>

jaxtonmaez, 2025-06-10:
My mistake. Here it is, I think.

2025-06-09T19:31:25.3507|INFO|PluginInstaller.cs|Install|91|Downloading plugin from https://bitbucket.org/NickHardy/nina.plugin.speckle/downloads/SpeckleInterferometry.2.2.1.3.zip
2025-06-09T19:31:26.0218|ERROR|PluginsVM.cs|InstallPlugin|139
System.Exception: Checksum of file C:\Users\educator\AppData\Local\Temp\Speckle Interferometry.dll does not match expected checksum B3D25EA96B8212CE7CE47C5BD25712763F44ED9A68D89102B27336917D2C505A of type SHA256. File may be corrupted
   at NINA.Plugin.PluginInstaller.Install(IPluginManifest manifest, Boolean update, CancellationToken ct) in C:\Projects\nina\NINA.Plugin\PluginInstaller.cs:line 117
   at NINA.ViewModel.Plugins.PluginsVM.<>c__DisplayClass10_0.<<InstallPlugin>b__0>d.MoveNext() in C:\Projects\nina\NINA\ViewModel\Plugins\PluginsVM.cs:line 135

marcblank, 2025-06-10:
Oh.  The problem is that the plugin author needs to update it.   The download was hosted at bitbucket, but that is no longer possible.   You’ll have to wait until that’s done

isbeorn, 2025-06-10:
@NickHardy As discussed last week, the plugin needs to be hosted somewhere else.

NickHardy, 2025-06-13:
Hi @jaxtonmaez 
I finally got round to fixing the plugin
Sorry, I was really busy last week.
Enjoy :)

closed 2025-06-13 by isbeorn