isbeorn/nina #12: Version NIGHTLY #032 throws ArgumentOutOfRangeException

insertnamehere1, 2025-06-11:
This exception was thrown when I was set up the Advanced Sequencer and dragging an dropping templates to to create a modified sequence. I’ve tried to reproduce for over an hour after the initial exception without success.

The log is attached.

This was not during an imaging session, no devices were attached. This was just “housekeeping”.

[20250611-014730-3.2.0.1032.9632-202506.log](https://github.com/user-attachments/files/20683336/20250611-014730-3.2.0.1032.9632-202506.log)

`2025-06-11T01:55:38.6675|ERROR|App.xaml.cs|Current_DispatcherUnhandledException|252|An unhandled exception has occurred of type System.ArgumentOutOfRangeException
2025-06-11T01:55:38.6686|ERROR|App.xaml.cs|Current_DispatcherUnhandledException|257
System.ArgumentOutOfRangeException: Specified argument was out of the range of valid values. (Parameter 'index')
   at System.Windows.Media.VisualCollection.Insert(Int32 index, Visual visual)
   at NINA.Sequencer.Behaviors.DragOverBehavior.AttachAdorner() in C:\Projects\nina\NINA.Sequencer\Behaviors\DragOverBehavior.cs:line 161
   at NINA.Sequencer.Behaviors.DragOverBehavior.MouseInObject(Object sender, MouseEventArgs e) in C:\Projects\nina\NINA.Sequencer\Behaviors\DragOverBehavior.cs:line 150
   at System.Windows.WeakEventManager.ListenerList`1.DeliverEvent(Object sender, EventArgs e, Type managerType)
`

This excerpt from the log points to the DragOverBehavior.cs file line 161 and the index variable being out of range. 
In AttachAdorner(), if layoutParent.Children does not have a DragDropAdorner, then the index value will remain at -1.  The insert at layoutParent.Children.Insert(index, dragOverAdorner) will throw the ArgumentOutOfRangeException.

I don't know why this is happening, but it appears to be.

Chris

marcblank, 2025-06-11:
Just a request that you not log at other than Info level unless asked.   These logs are tedious to review and rarely add value.

insertnamehere1, 2025-06-11:
Why discord? The problems I've logged are exceptions, not exactly "I need help" type problems. You want me to report these to discord?

insertnamehere1, 2025-06-11:
This is a very simple fix. I'm posting it here so you can add some defensive coding to handle the index that caused the issue.
I didn’t expect so many unrelated replies under this issue.

If you're suggesting Discord as a solution, you should also let the others posting here know about it.

marcblank, 2025-06-11:
Why did you close the issue?  I’m confused.

insertnamehere1, 2025-06-11:
Look, I'm appreciative of the work people have done on NINA and I'm using NIGHTLY builds to help identify bugs and in my own way try to help. I've lost hours of imaging time because NINA crash and I've spent many hours researching the exceptions I've found and in 3 cases tracked it down to the line of code that needs work. I've posted them on Bitbucket and now here. I know many other have spent way more time, but this is my attempt to help. I'm not doing this for me, I'm doing it for other users and developers to improve NINA.

So after posting the results of that work your following comments have been to turned debug off because it was "tedious" to read the logs and not to post here but in Discord. You told me this was better in Discord, that I should post there. So I closed this. You want me to put it in two places, I'm confused also.

I'm done.

insertnamehere1, 2025-06-11:
That will save time.

isbeorn, 2025-06-11:
Reopening as it's a very valid request. As you said it's a direct description of an issue with the necessary info included. No need for discord here.