isbeorn/nina #26: Plugin developers wishlist

JohannesWorks, 2025-07-15:
I want to reopen the topic here.
https://bitbucket.org/Isbeorn/nina/issues/1103/plugin-developers-wishlist 

My wish:
- Load sequences from files into the Advanced Sequencer

christian-photo, 2025-07-29:
I would like to mention, that even though it is possible via the API, I achieved it using reflection and I would prefer a solution that works without it. So this is still a wish on the wishlist

christian-photo, 2025-08-30:
These are still relevant for me:
- A method for getting the sequence root container from the sequence mediator
- Associate each SequenceEntity with an ID, that makes it possible to find the same item even if its values change
- Stopping a running auto focus