isbeorn/nina #32: 3.2 BETA1 - Startup sequence template should be placed in the "Sequence Start Area" of advanced sequencer

NGC1499, 2025-07-27:
[ X ] Is the issue reproducible?  
[ X ] Are you running the latest version?  
[ X ] Are all prerequisites that are mentioned inside the manual met?  

# Description #
Set a startup sequence template in Options -> Imaging. After restarting NINA, the startup instruction set appears in the  "Sequence Target Area" instead of "Sequence Start Area".

# Steps to Reproduce #
1. Go to Options -> Imaging
2. Set a template file for "Startup sequence template"
3. Restart NINA
4. Go to Advanced Sequencer and check the placement of the startup sequence instructions

## Expected behavior ##
The startup sequence instructions appear in the "Sequence Start Area".

## Actual behaviour ##
They're placed in the "Sequence Target Area".

# Version & Environment#
 - NINA : 3.2 Beta 1
 - Windows 11 24h2

marcblank, 2025-07-27:
This is as intended.  A startup sequence is a complete sequence.

marcblank, 2025-07-28:
The startup sequence is a full sequence loaded at NINA startup.  That full sequence can include any and all of the start, target, and end *areas*.   The start area isn’t a sequence.

NGC1499, 2025-07-29:
Thanks. I now understand the design. I was a bit misled by the caption "Startup".

Do you think it's better to rename the "Startup sequence template" to "Advanced sequence template"? This is also in par with "Legacy sequence template"... It's your call.

marcblank, 2025-07-29:
No that’s not right.  It’s a sequence that’s loaded at startup.

NGC1499, 2025-07-29:
It's fine. Issue is closed.

closed 2025-07-29 by NGC1499