isbeorn/nina #34: Wait for Dome Movement to Finish

kaito83, 2025-07-31:
Hi team,

I'm currently using a dome-controlled observatory with N.I.N.A., and I’ve run into a problem in the Advanced Sequencer: there is no built-in option to wait for the dome to complete its physical movement (e.g., rotation) before continuing to the next step of the sequence.

For instance, after a telescope slew or meridian flip, the dome starts rotating — but N.I.N.A. does not wait for the dome to finish moving before continuing to the next instructions (such as centering or imaging). This can result in imaging starting while the dome is still rotating and blocking the aperture.

There currently is no instruction like “Wait for Dome to Stop Moving”, and while it’s possible to insert a fixed Wait for Time Span block, this is not a reliable or elegant solution — especially if dome movement time varies due to acceleration or dome size.

Would it be possible to add an option in the Advanced Sequencer such as:

A “Wait for Dome Movement” instruction block

Or an internal check that pauses execution until IsSlewing = false or Slaved = true && Parked = false via ASCOM/Alpaca dome status

This would improve reliability, especially for remote or automated observatories.

Thanks for your amazing work and continued support of dome users!

Best regards,

kaito83, 2025-07-31:
Hi,

I'm not using ASCOM Device Hub — I'm connecting to all devices directly through N.I.N.A.
The N.I.N.A. UI correctly reports that the dome is slewing, but the Advanced Sequencer still proceeds to the next step without waiting for the movement to finish.

Just to mention: I'm using a custom firmware for the dome controller, but it properly updates the dome status and responds to ASCOM calls (e.g. Slewing, AtHome, etc.).

Here is an example from my ascom log:
```

10:48:58.693 Connected Get             True
10:48:58.693 Slewing                   True
10:48:58.693 Azimuth                   14,7044
10:48:58.693 Connected Get             True
10:48:58.693 Connected Get             True
10:48:58.694 Slewing                   True
10:48:58.812 UDP RX:                   f#1
10:48:58.867 UDP RX:                   c#1
10:48:58.918 UDP RX:                   D#14.9100
10:48:58.969 UDP RX:                   D#15.2688
10:48:59.119 UDP RX:                   D#15.6275
10:48:59.267 UDP RX:                   c#1
10:48:59.319 UDP RX:                   D#15.9862
10:48:59.468 UDP RX:                   D#16.3450
10:48:59.618 UDP RX:                   D#16.7038
10:48:59.695 Connected Get             True
10:48:59.695 Connected Get             True
10:48:59.695 ShutterStatus             shutterClosing
10:48:59.695 Connected Get             True
10:48:59.695 AtHome                    False
10:48:59.696 Connected Get             True
10:48:59.696 AtPark                    Threw an exception: 
ASCOM.PropertyNotImplementedException: Property read AtPark is not implemented in this driver.
   at ASCOM.EzDome.Dome.DomeHardware.get_AtPark()
   at ASCOM.EzDome.Dome.Dome.get_AtPark()
10:48:59.697 Connected Get             True
10:48:59.697 Slewing                   True
10:48:59.697 Azimuth                   16,7038
10:48:59.697 Connected Get             True
10:48:59.697 Connected Get             True
10:48:59.697 Slewing                   True
10:48:59.767 UDP RX:                   c#1
10:48:59.827 UDP RX:                   D#17.0625
10:48:59.968 UDP RX:                   D#17.4213
10:49:00.118 UDP RX:                   D#17.7800
10:49:00.268 UDP RX:                   c#1
10:49:00.318 UDP RX:                   D#18.1388
10:49:00.468 UDP RX:                   D#18.4975
10:49:00.618 UDP RX:                   D#18.8563
```

Thanks!

isbeorn, 2025-07-31:
Can you share the complete log? There are some ascom dome drivers out there that don't report slewing in-time which is not conforming. the full log might contain more hints of what's going on.

kaito83, 2025-07-31:
Hmm... thank you! I’ve made the changes in the ASCOM driver, I’ll test it.

clear skies!

closed 2025-08-01 by kaito83