isbeorn/nina #65: AbortExposure aborts the exposure but not the capture call

christian-photo, 2025-09-27:
**Describe the bug**
When calling `AbortExposure` on the `ICameraMediator` it will successfully cancel the exposure in the camera, however the capture call continues running. It looks like the exposure was not stopped in NINA and at the end it will try to download the exposure. After aborting the exposure it is already possible to start a new exposure. When this is done in NINA it works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Start an exposure
2. Abort the exposure (if using the api: `equipment/camera/abort-exposure`)
3. See how `IsExposing` changes to `false`, `CameraState` stays `Exposing` and the exposure continues

**Expected behavior**
It is expected, that the capture call is cancelled with the exposure

**Screenshots**

[image]

This is after aborting the exposure

**Desktop (please complete the following information):**
 - OS: Windows Version 10.0.26100 Build 26100
 - Version: 3.2 RC007 but the issue has been there for a longer time

marcblank, 2026-02-07:
Is this camera-specific?

christian-photo, 2026-02-08:
It doesn't seem to be, although it behaves differently for different cameras. Unfortunately I only have my Canon EOS 600d, the NINA simulator and the [Sky Simulator for ASCOM and ALPACA](https://sourceforge.net/projects/sky-simulator/) to test.

### Test setup
I created a sequence containing only a "Take Exposure" instruction with the time set to 10s. I am using my api to test abort exposure, which does nothing else than calling `AbortExposure` on the camera mediator. After starting I also get the Camera Info to compare it with the info after aborting the exposure.

### Canon EOS 600d
When the exposure starts `IsExposing` switches to `true`. Throughout the test `CameraState` doesn't change from `NoState`, but thats seems to be an issue with the camera and not related to this. After aborting the exposure `IsExposing` switches to `false` and a `Camera download failed` notification appears and the status bar is emptied. However the mirror continues to wait for the 10s to finish before going up again.

### NINA Simulator Camera
Here it is basically the same, however no notification appears and the status bar is also unaffected by calling abort-exposure. Basically, there is no indication of the success of calling abort exposure other than `IsExposing` switching to `false`.

### Sky Simulator (ALPACA)
After calling abort exposure, `IsExposure` changes to `false` but the `CameraState` stays `Exposing`. Again, there is no visual indication that the exposure has been aborted.

closed 2026-08-27 by isbeorn